uvic-auvic/Firmware-Motor_Controller_2.0 #4: Implement PWM in and out for all the timers that our connected

bates-andy123, 2017-09-21:
Implement a way to monitor the RPM and set the power level to each motor. This should be done use timers with DMA configured as the input. Store multiple time stamps from a timer and then do math to figure out the period. In versing the period will give the frequency which can be multiplied by 60 for RPM

lymacasm, 2018-03-08:
Do you think we could split this issue into two? We could have different files, one for PWN_in, and one for PWM_out. That would keep them independent, and I feel like the two tasks here are quite large on their own.

lymacasm, 2018-03-14:
I've made this the PWM in issue, and I will be making a PWM out issue shortly.

poornack, 2018-03-29:
Code for this issue has been completed and is ready for review. Pull request #19 has been created. Details about the code can be read in the pull request. Assigning this to Lyden review and merge to master.

poornack, 2018-05-08:
Code merged to master. Refer to PR #19 for more information

closed 2018-05-08 by poornack