uvic-auvic/Firmware-Motor_Controller_2.0 #6: Implement a task for reading the config pins

bates-andy123, 2017-09-21:
Write a task that reads the config pins into the micro controller. This tasks should execute before any other code on the micro-controller and store it result in some sort variable. Then it should boot up all regular tasks and then terminate itself. 

The idea is if we know the current state of the input pins we could use this task to change certain execution paths. Currently we have no use but could end up using it in the future.

poornack, 2018-05-18:
I'm assigning this to Eisa to finish up. The code pushed right now looks like it should work. Eisa will verify this with the motor controller board.

poornack, 2018-05-28:
Firmware has been verified with the Discovery board and merged to master. @shotbyric please test your code on the motor controller board next.