uvic-auvic/Firmware-Motor_Controller_2.0 #7: Get UART working

bates-andy123, 2017-10-20:
Making so UART can be implemented with our FSM Task

poornack, 2018-01-31:
Basic implementation of UART has been completed using interrupts. I was not able to get it done using DMA in time. I will continue working on UART with DMA as a lower priority project.

Some limitations right now:
- There is no check for buffer overflow
- If some command has a argument that equals the value of the new line character, the UART code will detect that as the end of the command. Current plan for this is to have two end of command characters, '\n\r'.

lymacasm, 2018-01-31:
I wonder if it would maybe make more sense to make it all ASCII? Like, a command would be M1F85\n, that is 'M' '1' 'F' '8' '5' '\n'? Because, our end of command terminator only works if you assume we stick with ascii. As soon as we move away from ASCII, then the new line command terminator doesn't quite work anymore.

lymacasm, 2018-01-31:
We could make the speed between 00 and 99. So that our command would be 5 characters long, and if you wanted a speed less than 10, you'd need to send the leading '0' character.

poornack, 2018-01-31:
That's a good idea Lyden. I like this over using \n\r. The current UART code will handle this scenario. FSM.c will need to be changed to handle this when that gets done. The maximum command length is 8 bytes so if you need for higher precision for the motor speed, the command can be extended to 3 bytes for the speed without any problem.

lymacasm, 2018-01-31:
Awesome, it shouldn't be too hard to change in FSM! I think we should go with that moving forward!

poornack, 2018-02-28:
Implementation of UART using interrupts is complete.

The output buffer size is set to 64 bytes for now. This can be changed if necessary. The function to write data through UART will return 1 if the data was added to the output buffer successfully, -1 if the length of the string to write is zero and -2 if the buffer will overflow if the given string is written to the output buffer.

I've updated the Wiki for this repo with the command structure that was agreed upon in the above messages. I will update the fsm.c file to reflect this as well when I start working on the issue for the FSM.

Assigning this to Lyden to merge to code to master.

lymacasm, 2018-02-28:
Changes have been merged with master.

closed 2018-02-28 by lymacasm