uvic-auvic/Firmware-Motor_Controller_2.0 #12: Make the ADC driver

bates-andy123, 2018-03-08:
Design a low level driver to measure the multiplexers and the pressure sensors 

This driver can either be continuous or event driven. It is up to you

lymacasm, 2018-03-08:
Would #8 not depend on this? And should this maybe be done as a part of #8? Or would this be a more general driver for more than just the multiplexer ADC's?

bates-andy123, 2018-03-08:
Will it needs be integrated with #8  but it also has to do the water sensor.  I think the best description would be currently blocked by #8. 

The idea is to setup the entire ADC as they all use the same one. So we need to consider everything at once.

lymacasm, 2018-03-08:
That makes sense. Good plan!

bates-andy123, 2018-05-16:
The ADC driver should be able to read the water sensor and all the current and temperature sensors on the motors. Save these results to an array.

Using DMA is not recommend, instead use interrupts. This is because a multiplexer is between between the analog voltages of the motors and the ADC on the micro.

Make sure to average the results. An average of 5 would be satisfactory.

closed 2018-06-12 by rkeen9