uvic-auvic/Firmware-Motor_Controller_2.0 #27: Implement a system to reset RPM to zero if no edge is detected in a while

bates-andy123, 2018-05-17:
One of the current problems is that RPM will stay at whatever value is last read. For example if the last read RPM is 7000 and the motor_controller does not see any more edges then the RPM will stay at 7000. 

Our system should set RPM to 0 if it does not see any edges in a certain amount of time.

bates-andy123, 2018-05-17:
After talking to Poorna we think the easiest way to do this is by monitoring the current out of the motor controller. If it is below 20mA above just the ESC draw then we will set RPM to off. Waiting for issue #12

lymacasm, 2018-05-17:
Good idea!

lymacasm, 2018-07-26:
Instead of doing this, both the index of the last data transfer, and the contents of the data transfer are investigated. If they have not changed, then the RPM for that motor should be set to 0. See pull request #37 .

PR #37 (2018-07-26): RPM's will now reset after a certain period of time
This pull request closes #27 .

Added an RTOS task whose purpose is to monitor the RPM's and set the RPM to 0 if no DMA event has happened since the last check. Currently, will check once every 300ms, meaning the minimum RPM will ideally be 200rpm. One disadvantage with this method is that if the RPM is updated immediately after checking, the longest period a channel could go without being reset is 600ms (in the extreme case), which would correlate to an RPM of 100rpm. I don't think this is going to be an issue though, because it's an unlikel [... cut]
closed 2018-07-26 by lymacasm